stm32duino/Arduino_Core_STM32 #2585: I2C Slave does not work (STM32F103)

AndKe, 2024-12-01:
**Describe the bug**
I2C slave does not recieve requests.

**To Reproduce**

```
#include <Wire.h>
long lastms;
volatile int x = 0;
volatile int y = 0;

static void receiveEvent(int howMany) {
  x = Wire.read();
}

static void requestEvent() {
  y = 1;
}

void setup() {
  Serial.begin(115200);

  pinMode(PB6, INPUT); //PB6=SCL
  pinMode(PB7, INPUT); //PB7=setSDA
  pinMode(PC13, OUTPUT);

  Wire.setSCL(PB6);
  Wire.setSDA(PB7);
  Wire.begin(11);                // join i2c bus as slave
  //Wire.onRequest(requestEvent);  // register event
  Wire.onReceive(receiveEvent);  // register event
  lastms = millis();
}

void loop() {
  if (millis() > lastms + 1000) {
    Serial.print("-");
    lastms = millis();

  }
  if (x != 0) {
    Serial.print("U:0x");
    Serial.println(x, HEX);
    //digitalWrite(PC13,!digitalRead(PC13));
    x = 0;
  }
  if (y != 0) {
    Serial.println("V");
    y = 0;
  }
}

```

Steps to reproduce the behavior:
1. Run the code om STM32F103Cx
2. the code expects requests from master to address x0xB / 11d
3. I see the data requests using a oscolloscope/logic analyzer on PB6/PB7
4. The software outputs only "-"

**Expected behavior**
I'd expect it to work like it does on Atmega/atmel controllers - where this code works.   see https://github.com/arduino/ArduinoCore-API/issues/241

Than I rebuilt it all using Arduino Core 2.9.0  (current release)
On Core 2.9.0 - the PB6 is at least not pulled low - which is good, but it still does not work.

[image]

**Desktop (please complete the following information):**
 - OS: Linux Ubuntu24.10
 - Arduino IDE version: 2.3.3
 - STM32 core version: 2.9.0
 - Tools menu settings if not the default: [e.g. Newlib Standard, No Serial]
 - Upload method: SWD

**Board (please complete the following information):**
 - Name: Bluepill STM32F103C8
 - Hardware Revision: n/a
 - Extra hardware used if any: -

**Additional context**
Add any other context about the problem here.

fpistm, 2024-12-01:
Do you have pullup resistors on each I2c lines?

AndKe, 2024-12-01:
Yes, and - I can see the I2C packets using an oscilloscope on the pins. (decoded on the scope)  
Plus : for simple troubleshooting I hooked it up on a breadboard. When I do the very same setup, but switch the blackpill an Atmega based arduino boards - it works.

fpistm, 2024-12-01:
Could you share the master example?

AndKe, 2024-12-01:
@fpistm ubable, the master is Ardupilot (which is open source)- if you have some hardware that can run Ardupilot, and configure BATT_MONITOR 5  - then you have that I2C traffic too.

AndKe, 2024-12-02:
Please see attached communication capture with device 0xB    - the capture should clarify any questions you might have about protocol/timing etc. it is captured using Saleae logic analyzer, which is perfectly multi-platform with nice software: https://www.saleae.com/pages/downloads

[BMSEmu Boot.zip](https://github.com/user-attachments/files/17973614/BMSEmu.Boot.zip)

fpistm, 2024-12-02:
Thanks for inputs. I will try to test on my side but don't know when as I'm fully busy on other tasks.

AndKe, 2024-12-02:
Thank you, I am really stuck on this one, I "modernized" previous design (Atmega32u4) by dropping in STM32F103, and now - with plenty of new PCB's manufactured, I am stuck with this issue.
I have previously had projects with STM32F103C8 and CB  acting as master without any issues.

AndKe, 2024-12-02:
Additional information; the exactly same happens when SCL=PB10 and SDA=PB11

AndKe, 2024-12-02:
@fpistm I understand that you are busy, but is there some way to ...motivate... you or another competent contributor here to solve this?

AndKe, 2024-12-03:
Yes, the blackpill is an error on my side.
The device in question is something that did not work as I2C slave after being ported from Atmega32U4  And all issues here are about STM32F103C8T6.  be it on my hardware, or Bluepill.

After much troubleshooting I made this test app above, and ran it on Atmega32u4 for comparsion/to verify it.
Then I also moved from my board to an actual bluepill (just to easily test on PB10/PB11) .
During the compilations for various targets, I must have messed it up.
I will retest that once I get home, in about 2 hours.

BTW: I have also tried to build it on VSCode w/platformio (and Arduino library)  - with same results. So me trying both Arduino and all that are kind of desperate attempts to make it work :)

ItsXor, 2024-12-04:
I can confirm this issue exists.
Only input I can give is that Wire Slave sometimes accepts data right after reset, maybe first 300ms.
This issue is not present in other core (https://github.com/rogerclarkmelbourne/Arduino_STM32).
Would love this to be resolved because SimpleFOC only works with this core, but I can't control my Bluepill as I2C slave simultaneously.

AndKe, 2024-12-04:
@ItsXor Thank you! - for making me doubt my sanity.  I started to wonder how this was possible, without anyone spotting it before.
I would love to know what can be brought over from that core into PlatformIO  (as my project is really there, and I do not really know if PlatformIO can somehow use that core without much hacking.
On the other hand, I just hope one of the skilled STM engineers can trow together a fix soon.

AndKe, 2024-12-07:
@fpistm  Dear STM - please note: "all I want for Christmas is a fix for I2C Slave"   - this is the losest "writing to Santa" i ever did.  :)

AndKe, 2024-12-12:
@fpistm very glad to hear that. Unfortunately I won't be of much help nailing the issue(insufficient skills in the area), but I definitely can help testing or running a debug build.

fpistm, 2024-12-17:
@AndKe 
I've start to check your issue.
What is exactly your issue? 
Your talk about

fpistm, 2024-12-17:
> 2- the code expects requests from master to address x0xB / 11d

but `onRequest` is comment on your code and see no `Wire.write` on your code?
So I don't understand your issue.

fpistm, 2024-12-17:
Here test sketch:

* <details>
  <summary>Master</summary>

```c++
#include <Wire.h>

#define I2C_ADDR  11

static byte x = 1;

void setup()
{
  Wire.begin();        // join i2c bus (address optional for master)
  Serial.begin(9600);  // start serial for output
}

void loop()
{
  Wire.requestFrom(I2C_ADDR, 1);  // request 1 bytes from slave device

  while(Wire.available())         // slave may send less than requested
  {
    char c = Wire.read();         // receive a byte as character
    Serial.println(c, HEX);              // print the character
  }

  delay(10);

  Wire.beginTransmission(I2C_ADDR); // transmit to device
  Wire.write(x);                    // sends one byte
  Wire.endTransmission();           // stop transmitting
  x++;

  delay(500);
}
```
</details>

* <details>
  <summary>Slave based on your sketch</summary>

```c++
#include <Wire.h>
long lastms;
volatile int x = 0;
volatile int y = 0;

static void receiveEvent(int howMany) {
  x = Wire.read();
}

static void requestEvent() {
  Wire.write(y++);
}

void setup() {
  Serial.begin(115200);
  Wire.begin(11);                // join i2c bus as slave
  Wire.onRequest(requestEvent);  // register event
  Wire.onReceive(receiveEvent);  // register event
  lastms = millis();
}

void loop() {
  if (millis() > lastms + 1000) {
    Serial.print("-");
    lastms = millis();

  }
  if (x != 0) {
    Serial.printf("U:0x %x\n", x);
    x = 0;
  }
}
```
</details>

AndKe, 2024-12-17:
@fpistm I believe that the commented `Wire.onRequest(requestEvent)` part was just another desperate test.
The main issue is that when acting as slave, (recieving traffic) the device just kept holding SCL low.

I uploaded the "Slave based on your sketch" that you posted,
On the bus, I have a ArduPilot I2C as master.  it is requesting data: 
[image]

Please note how the SCL is suddenly kept low indefinitely.

Then another capture: the F103 is running fine for several selconds, and then I connect the I2C bus (where the master is talking to it.)
[image]

Please note that in both cases: the F103 is the one that is stuck keeping the SCL low - intil reset - disconnecting master leaves a fully functional I2C master.

Please see the Saleae logic capture - you can open the files using https://www.saleae.com/pages/downloads  the tool is multiplatform and free.

[Saleae capture.zip](https://github.com/user-attachments/files/18169806/Saleae.capture.zip)

fpistm, 2024-12-17:
Please test sketches provided.
I can't reproduce.

AndKe, 2024-12-17:
@fpistm  
Ardupilot project: https://github.com/ArduPilot/ardupilot
It uses I2C quite much for all kinds of internal sensors, but usually has an external I2C bus for less-critical devices.
To be precise: this is the part I am trying to communicate with: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_BattMonitor/AP_BattMonitor_SMBus_Solo.cpp

Yes, I've been emulating that BMS for several years using Atmega 328 and Atmega32u4  with Arduino libraries. 
This whole issue started as I discovered that compiling my code for STM32F103 
(I have used STM32 for many other projects, as I2C master,  and it always worked fine)

That's why we are now trying to do basic stuff like trying to see a write requests as a slave. 

I loaded master sketch on another STM32F103 - but it does not generate I2C traffic. (both lines remain high)

Here is my slave firmware:
[I2C_test_slave.ino.zip](https://github.com/user-attachments/files/18170866/I2C_test_slave.ino.zip)
And my master firmware:
[I2C-test-master.ino.zip](https://github.com/user-attachments/files/18170876/I2C-test-master.ino.zip)

Maybe there's something with the toolchain/build environment?
The fact is that I first had trouble with this in VSCode + PlatformIO , then tried Arduino IDE (now 2.3.4)  - and I see no difference. That's why I thought this is related to the wire library.

fpistm, 2024-12-17:
With pullup on each I2C lines?

fpistm, 2024-12-17:
Usually 4.7k and stm32 operate at 3.3v even if gpio are 5v tolerant.

AndKe, 2024-12-17:
I know, but this is not the issue here, I do not see some odd levels when looking at the signals with an oscilloscope, crisp and clear  - (when there is something to see)

fpistm, 2024-12-18:
I've tested your slave binary and it works on a BluePillF103C8 (with a guenine STM32F103C8) and it works.

> does your master-build cause any I2C traffic?
As stated it works! 
My setup is simple 2 boards with I2C lines wired (with 4.7K PU on each I2C lines), GND interconnected and that's it.

Request from the master:
[image]
then 10 ms later sent from the Master:
[image]

fpistm, 2024-12-18:
Finally I've tested with an Ardunio UNO as master (5V) and BluePill with your slave binary and it works.