stm32duino/Arduino_Core_STM32 #2589: [New Variant] STM32F407IGH

hlgzhx, 2024-12-04:
**Board description (please complete the following information):**
 - Name: Generic STM32F407IGH

I‘ve found the variant folder of F407I(E-G)(H-T)_F417I(E-G)(H-T) in "STMicroelectronics\hardware\stm32\2.8.1\variants\STM32F4xx",but it is not added in the "STMicroelectronics\hardware\stm32\2.8.1\boards.txt".
After I add:
# Generic F407IGHx
GenF4.menu.pnum.GENERIC_F407IGHX=Generic F407IGHx
GenF4.menu.pnum.GENERIC_F407IGHX.upload.maximum_size=524288
GenF4.menu.pnum.GENERIC_F407IGHX.upload.maximum_data_size=131072
GenF4.menu.pnum.GENERIC_F407IGHX.build.board=GENERIC_F407IGHX
GenF4.menu.pnum.GENERIC_F407IGHX.build.product_line=STM32F407xx
GenF4.menu.pnum.GENERIC_F407IGHX.build.variant=STM32F4xx/F407I(E-G)(H-T)_F417I(E-G)(H-T)
GenF4.menu.pnum.GENERIC_F407IGHX.debug.svd_file={runtime.tools.STM32_SVD.path}/svd/STM32F4xx/STM32F407.svd

in it,and put the ldscript.ld in F407I(E-G)(H-T)_F417I(E-G)(H-T) folder,it works perfectly.i tested several functions include Can bus.They all work correctly.My board is Robomaster Development Board Mk. C whose Mcu is STM32F407IGH.
May you stuffs add this beforehead during next version?thanks very much .

fpistm, 2024-12-12:
Hi @hlgzhx 

did you plan to provide a PR ?

closed 2024-12-17 by fpistm

hlgzhx, 2024-12-18:
sorry i forgot to answer that.I'd like to provide a PR

fpistm, 2024-12-18:
No worry.  This issue is referenced. When you have time to do the PR it will be linked to this issue and update accordingly.

PR #2649 (2025-02-05): add support for stm32F407IGHx
**Summary**

add support for Generic STM32F407IGHx boards by editing boards.txt and add variants/STM32F4xx/F407I(E-G)(H-T)_F417I(E-G)(H-T)/ldscript.ld

[image]
compile test is passed,my board is in school but school is closed.if you need further test on uploading,please wait me to get back to school after Chinese new year.

im glad to fix further issues on supporting stm32f407ighx boards

Fixes #2589.
Fixes #2131.
commit 3510e4c9 (2025-02-05): variant(f4): add generic F407I(E-G)(H-T) and F417I(E-G)(H-T)

Fixes #2131.
Fixes #2589.