stm32duino/Arduino_Core_STM32 #2590: Uploading Sketch with OpenOCD STLink (SWD) is messing up the path and fails

momonator25, 2024-12-04:
**To Reproduce**

Steps to reproduce the behavior:
1. open any example e.g. "Blink.ino"
2. config the board settings as listed below
3. hit the "Upload" Button
4. wait for the sketch to compile
5. upload fails because of path issue (mixed file separators "/" and "\" are used as arguments when calling OpenOCD. "\" are interpreted as escape characters and mess up the path)

```powershell
Sketch uses 13616 bytes (10%) of program storage space. Maximum is 131072 bytes.
Global variables use 1224 bytes (9%) of dynamic memory, leaving 11064 bytes for local variables. Maximum is 12288 bytes.
"C:\Users\Vierneusel1\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-openocd\0.12.0-4/bin/openocd.exe" -d2 -f interface/stlink.cfg -f target/stm32wbx.cfg -c "program C:\Users\Vierneusel1\AppData\Local\arduino\sketches\A63CD47ED22449C201556EDFD2873119/Blink.ino.elf verify reset exit"
xPack Open On-Chip Debugger 0.12.0+dev-01685-gb9224c0c0-dirty (2024-08-02-19:51)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 500 kHz
Info : STLINK V2J35S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.244061
Info : [stm32wbx.cpu] Cortex-M4 r0p1 processor detected
Info : [stm32wbx.cpu] target has 6 breakpoints, 4 watchpoints
Info : [stm32wbx.cpu] Examination succeed
Info : [stm32wbx.cpu] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
[stm32wbx.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08002994 msp: 0x20003000
** Programming Started **
Error: couldn't open C:UsersVierneusel1AppDataLocalrduinosketchesA63CD47ED22449C201556EDFD2873119/Blink.ino.elf
** Programming Failed **
shutdown command invoked

Failed uploading: uploading error: exit status 1
```

**Expected behavior**
Running the command with fixed path variable of the sketch from powershell directly results in expected bahavior:

```powershell
PS M:\> &"C:\Users\Vierneusel1\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-openocd\0.12.0-4/bin/openocd.exe" -d2 -f interface/stlink.cfg -f target/stm32wbx.cfg -c "program C:/Users/Vierneusel1/AppData/Local/arduino/sketches/A63CD47ED22449C201556EDFD2873119/Blink.ino.elf verify reset exit"
xPack Open On-Chip Debugger 0.12.0+dev-01685-gb9224c0c0-dirty (2024-08-02-19:51)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 500 kHz
Info : STLINK V2J35S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.244061
Info : [stm32wbx.cpu] Cortex-M4 r0p1 processor detected
Info : [stm32wbx.cpu] target has 6 breakpoints, 4 watchpoints
Info : [stm32wbx.cpu] Examination succeed
Info : [stm32wbx.cpu] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
[stm32wbx.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08002994 msp: 0x20003000
** Programming Started **
Info : device idcode = 0x20016495 (STM32WB5x - Rev 2.1 : 0x2001)
Info : RDP level 0 (0xAA)
Info : flash size = 1024 KiB
Info : flash mode : single-bank
Info : Padding image section 1 at 0x08003684 with 4 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x08003688 .. 0x08003fff
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
shutdown command invoked
```

But only using Mac/Unix seperator like above works!

This also fails:

```powershell
PS M:\> &"C:\Users\Vierneusel1\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-openocd\0.12.0-4/bin/openocd.exe" -d2 -f interface/stlink.cfg -f target/stm32wbx.cfg -c "program C:\Users\Vierneusel1\AppData\Local\arduino\sketches\A63CD47ED22449C201556EDFD2873119\Blink.ino.elf verify reset exit"
```

**Desktop:**
 - OS: Windows 11
 - Arduino IDE version: 2.3.4 (fresh install)
 - STM32 core version: 2.9
 - Tools menu settings: all default
 - Upload method: OpenOCD STLink SWD

**Board:**
 - Name: Generic WB55CGUx
 - Hardware Revision: WeActStudio.STM32WB55CoreBoard

fpistm, 2024-12-04:
Hi @momonator25 
Thanks for pointing this.

> Where the `{build.path}` is given to the upload command as a variable and I guess it comes from the IDE?

You're right so it will not be easy to fix as it is Arduino IDE which should handle this. Anyway, I guess openocd should protect this also but do not.... 😩

fpistm, 2024-12-04:
@momonator25 
I've found the right way to protect the path and made a PR #2591 
Here the patch in `platform.txt`:
```patch
-tools.openocd_upload.upload.pattern="{path}/{cmd}" {upload.verbose} -f interface/{upload.protocol}.cfg -f target/{openocd.target}.cfg -c "program {build.path}/{build.project_name}.elf verify reset exit"
+tools.openocd_upload.upload.pattern="{path}/{cmd}" {upload.verbose} -f interface/{upload.protocol}.cfg -f target/{openocd.target}.cfg -c "program {{build.path}/{build.project_name}.elf} verify reset exit"
```

Could you test it on your side? On my side it work on Windows and Linux.

momonator25, 2024-12-04:
> @momonator25 I've found the right way to protect the path and made a PR #2591 Here the patch in `platform.txt`:
> 
> ```diff
> -tools.openocd_upload.upload.pattern="{path}/{cmd}" {upload.verbose} -f interface/{upload.protocol}.cfg -f target/{openocd.target}.cfg -c "program {build.path}/{build.project_name}.elf verify reset exit"
> +tools.openocd_upload.upload.pattern="{path}/{cmd}" {upload.verbose} -f interface/{upload.protocol}.cfg -f target/{openocd.target}.cfg -c "program {{build.path}/{build.project_name}.elf} verify reset exit"
> ```
> 
> Could you test it on your side? On my side it work on Windows and Linux.

Will try tomorrow on my laptop and reply here with the result.

momonator25, 2024-12-04:
> @momonator25 I've found the right way to protect the path and made a PR #2591 Here the patch in `platform.txt`:
>
> ```diff
> -tools.openocd_upload.upload.pattern="{path}/{cmd}" {upload.verbose} -f [...]

It does work with the fix! Thanks for the fast response :)

commit ffb23d6b (2024-12-04): fix(openocd): path on windows

Fixes #2590
closed 2024-12-04 by momonator25