stm32duino/Arduino_Core_STM32 #2605: STM32G474CEU6 for WeAct Boards

imk2021, 2024-12-13:
Hello,

The wonderful WeAct company to be found on AliExpress and board designs to be found on this wonderful site.
Have introduced two new boards based on the STM32G474CEU6   UFQFPN48
https://www.aliexpress.com/item/1005007412335651.html?spm=a2g0o.order_list.order_list_main.31.76b71802BckICR

https://www.aliexpress.com/item/1005006170922820.html?pdp_npi=4%40dis%21GBP%21%EF%BF%A14.52%21%EF%BF%A14.12%21%21%2140.51%2136.92%21%402103849717341275918568641ef4ef%2112000037359413670%21sh%21UK%212641238368%21X&spm=a2g0o.store_pc_allItems_or_groupList.new_all_items_2007524327660.1005006170922820

I would like to make a request to add these boards into STM32 Arduino system please.

Many thanks in advance IMK

**Board description (please complete the following information):**
 - Name: [e.g. DISCOVERY F769NI]
 - Link to all relevant information:
   * Documentations
   * Schematics
   * Where this hardware can be purchased
  * ...

**Note that user can add a STM32 based board following this [wiki](https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-(board))**

Any contribution is welcome, so do not hesitate to submit a PR.

All new variant request will be moved to [New Variant] request list #722 and the related issue will be closed.

fpistm, 2024-12-14:
Added to #722

ALTracer, 2024-12-26:
Dear maintainers,
I have written a +200 lines patch adding support for this board (of which I have a couple to test on) against 2.6.0 branch. I can contribute it to 2.9.0 branch or latest master. Would you be interested in taking it?

fpistm, 2024-12-26:
Hi @ALTracer 
Of course. Feel free to open a PR.

imk2021, 2024-12-26:
Hello ALTracer
I have two of these WeAct STM32G474CEU6 and happy to test your variant board support when it is published.
IMK

closed 2025-01-04 by fpistm