stm32duino/Arduino_Core_STM32 #2616: Serial1 undefined for STM32F103C8Tx.

AmirReza-Bahrami, 2024-12-28:
hello. Serial1 undefined for STM32F103C8Tx. Meanwhile, Serial1 is defined for bluepill f103c8 whose microcontroller is exactly STM32F103C8T6.

fpistm, 2024-12-28:
Generic defined default `Serial` as `Serial2`:
https://github.com/stm32duino/Arduino_Core_STM32/blame/bb640bfaa12813f8674f7d52e7cf4a9cc52ae149/variants/STM32F1xx/F103C8T_F103CB(T-U)/variant_generic.h#L135

You have to define it or redefine it as described by the wiki.

closed 2024-12-28 by fpistm

AmirReza-Bahrami, 2024-12-28:
> Generic defined default `Serial` as `Serial2`: https://github.com/stm32duino/Arduino_Core_STM32/blame/bb640bfaa12813f8674f7d52e7cf4a9cc52ae149/variants/STM32F1xx/F103C8T_F103CB(T-U)/variant_generic.h#L135
> 
> You have to define it or redefine it as described by the wiki.

The problem is that the Serial1 macro is also undefined. If I want to use all three u(s)art ports, I will have a problem. In bluepill stm32f103c8 this problem does not exist and all u(s)art ports (Serialusb, Serial1, Serial2, Serial3) are defined but in stm32f103c8tx Serial1 is undefined. For example Serial1.begin(9600); It gives an error.

fpistm, 2024-12-28:
You have to instantiate it.

AmirReza-Bahrami, 2024-12-28:
> You have to instantiate it.
> 
> https://github.com/stm32duino/Arduino_Core_STM32/wiki/API#hardwareserial

Thanks.