stm32duino/Arduino_Core_STM32 #2620: The Windows path for the OpenOCD STLink (SWD) upload method doesn't seem to be formatted correctly

jferdelyi, 2025-01-07:
**Describe the bug**
I'm sorry if this isn't the right place to report this, I'm relatively new to the STM world.
The Windows path for the OpenOCD STLink (SWD) upload method doesn't seem to be formatted correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Plug the board
2. Setup IDE: board, variant and upload method
3. Upload a sketch
4. See error

```
...
"C:\Users\jferdelyi\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-openocd\0.12.0-4/bin/openocd.exe" -d2 -f interface/stlink.cfg -f target/stm32f4x.cfg -c "program C:\Users\jferdelyi\AppData\Local\arduino\sketches\FFB7D34698BE1D7E261D15C786251A94/sketch_jan7a.ino.elf verify reset exit"
...
Error: couldn't open C:UsersjferdelyiAppDataLocalrduinosketchesFFB7D34698BE1D7E261D15C786251A94/sketch_jan7a.ino.elf
```
The command work well on CMD using this format
```
"C:\Users\jferdelyi\AppData\Local\Arduino15\packages\STMicroelectronics\tools\xpack-openocd\0.12.0-4/bin/openocd.exe" -d2 -f interface/stlink.cfg -f target/stm32f4x.cfg -c "program C:\\Users\\jferdelyi\\AppData\\Local\\arduino\\sketches\\FFB7D34698BE1D7E261D15C786251A94/sketch_jan7a.ino.elf verify reset exit"
```

**Expected behavior**
Path of my skecth should probably be: C:\\Users\\jferdelyi\\AppDataLocal\\arduino\\sketches\\FFB7D34698BE1D7E261D15C786251A94/sketch_jan7a.ino.elf

**Desktop:**
 - OS: Windows 11
 - Arduino IDE version: 2.3.4
 - STM32 core version: 2.9.0
 - Upload method: OpenOCD STLink (SWD)

**Board:**
 - Name: Nucleo64-F446RE
 - Hardware Revision: IDK (maybe Rev C)

jferdelyi, 2025-01-07:
Ho thank you, sorry for that.

closed 2025-01-07 by jferdelyi

fpistm, 2025-01-07:
De rien @jferdelyi 😉