stm32duino/Arduino_Core_STM32 #2653: [New Variant] STM32WB55VGY

federicaventriglia, 2025-02-10:
- Name: STM32WB55VGY
 - Link to all relevant information:
https://www.st.com/en/microcontrollers-microprocessors/stm32wb55vg.html

I can see there's a folder in the stm32 core `WB55V(C-E-G)(Q-Y)_WB55VYY` but it doesn't show as an in the IDE?

federicaventriglia, 2025-02-10:
I think I was able to answer everything myself, using the existing information on the wiki.

federicaventriglia, 2025-02-11:
I've tried to add support for the _STM32WB55VGYx_ MCUs, generating the ldscript and using a custom clock. I don't believe it's correct but it does compile, but my arduino sketch (Blink.ino) with a GPIO for an LED isn't behaving appropriately.

In my ldscript.ld I've modified the memory area like this 

```
/* Specify the memory areas */
MEMORY
{
FLASH (rx)                 : ORIGIN = 0x08000000, LENGTH = LD_MAX_SIZE - LD_FLASH_OFFSET
RAM1 (xrw)                 : ORIGIN = 0x20000008, LENGTH = LD_MAX_DATA_SIZE - 8
RAM_SHARED (xrw)           : ORIGIN = 0x20030000, LENGTH = 10K
}
```

and for the SystemClock, I've imported it from cube, according to all the peripherals I intended to use.

[WB55V(C-E-G)(Q-Y)_WB55VYY.zip](https://github.com/user-attachments/files/18752176/WB55V.C-E-G.Q-Y._WB55VYY.zip)

fpistm, 2025-02-12:
Hi @federicaventriglia 
did you try to debug to see if clock config is correct.

fpistm, 2025-02-12:
One issue in the `ldscript.ld`:
```patch
- FLASH (rx)                 : ORIGIN = 0x08000000, LENGTH = LD_MAX_SIZE - LD_FLASH_OFFSET
+ FLASH (rx)                 : ORIGIN = 0x08000000 + LD_FLASH_OFFSET, LENGTH = LD_MAX_SIZE - LD_FLASH_OFFSET
```

In the generic clock, pay attention to the lock ressource management, see other WB variant:
  
```
#include "lock_resource.h"
  /* This prevents concurrent access to RCC registers by CPU2 (M0+) */
  hsem_lock(CFG_HW_RCC_SEMID, HSEM_LOCK_DEFAULT_RETRY);
  /* This prevents the CPU2 (M0+) to disable the HSI48 oscillator */
  hsem_lock(CFG_HW_CLK48_CONFIG_SEMID, HSEM_LOCK_DEFAULT_RETRY);
```

As said if you try to debug, you will see if you end in an `Error_Handler()` during the clock config.

federicaventriglia, 2025-02-12:
Thanks for catching the issue in the ldscript, I've modified it now:

```
/* Specify the memory areas */
MEMORY
{
FLASH (rx)                 : ORIGIN = 0x08000000 + LD_FLASH_OFFSET, LENGTH = LD_MAX_SIZE - LD_FLASH_OFFSET
RAM1 (xrw)                 : ORIGIN = 0x20000008, LENGTH = LD_MAX_DATA_SIZE - 8
RAM_SHARED (xrw)           : ORIGIN = 0x20030000, LENGTH = 10K
}
```

As for the Clock config, I have also tried adding the lock/unlock, as you said I've noticed it from the P-NUCLEO implementation and I see the same behaviour. Not sure if I've used it properly but I'll attach the code as well. 

(I've also tried removing the custom clock configuration and using the default one (leaving the systemclock_config method empty) and I still see the same issue happening, which is strange)

Custom Clock with Locks

```
#if defined(ARDUINO_GENERIC_WB55VCQX) || defined(ARDUINO_GENERIC_WB55VCYX) ||\
    defined(ARDUINO_GENERIC_WB55VEQX) || defined(ARDUINO_GENERIC_WB55VEYX) ||\
    defined(ARDUINO_GENERIC_WB55VGQX) || defined(ARDUINO_GENERIC_WB55VGYX) ||\
    defined(ARDUINO_GENERIC_WB55VYYX)
#include "pins_arduino.h"
#include "lock_resource.h"

/**
  * @brief  System Clock Configuration
  * @param  None
  * @retval None
  */
WEAK void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {};

 /* This prevents concurrent access to RCC registers by CPU2 (M0+) */
  hsem_lock(CFG_HW_RCC_SEMID, HSEM_LOCK_DEFAULT_RETRY);
  /* This prevents the CPU2 (M0+) to disable the HSI48 oscillator */
  hsem_lock(CFG_HW_CLK48_CONFIG_SEMID, HSEM_LOCK_DEFAULT_RETRY);

  /** Configure the main internal regulator output voltage
  */
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);

  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI48|RCC_OSCILLATORTYPE_HSI
                              |RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_ON;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSI48State = RCC_HSI48_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLM = RCC_PLLM_DIV8;
  RCC_OscInitStruct.PLL.PLLN = 32;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLR = RCC_PLLR_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = RCC_PLLQ_DIV2;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
    #error "Clock Error Handler 1!"
  }

  /** Configure the SYSCLKSource, HCLK, PCLK1 and PCLK2 clocks dividers
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK4|RCC_CLOCKTYPE_HCLK2
                              |RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.AHBCLK2Divider = RCC_SYSCLK_DIV2;
  RCC_ClkInitStruct.AHBCLK4Divider = RCC_SYSCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_3) != HAL_OK)
  {
    Error_Handler();
    #error "Clock Error Handler 2!"
  }

  RCC_PeriphCLKInitTypeDef PeriphClkInitStruct = {};

  /** Initializes the peripherals clock
  */
  PeriphClkInitStruct.PeriphClockSelection = RCC_PERIPHCLK_SMPS;
  PeriphClkInitStruct.SmpsClockSelection = RCC_SMPSCLKSOURCE_HSI;
  PeriphClkInitStruct.SmpsDivSelection = RCC_SMPSCLKDIV_RANGE1;

  if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct) != HAL_OK)
  {
    Error_Handler();
    #error "Clock Error Handler 3!"
  }

  hsem_unlock(CFG_HW_RCC_SEMID);
  hsem_unlock(CFG_HW_CLK48_CONFIG_SEMID);

}

#endif /* ARDUINO_GENERIC_* */

```

I'm not too familiar with the debugging process so forgive me I'm not doing it properly. I've added some error statements in the Error_Handler() calls, would that work?

federicaventriglia, 2025-02-12:
> Thanks for catching the issue in the ldscript, I've modified it now:
>
> ```
> /* Specify the memory areas */
> MEMORY
> {
> FLASH (rx)                 : ORIGIN = 0x08000000 + LD_FLASH_OFFSET, LENGTH = [...]

Seems like the same thing is happening with default clock config. 

I've also tried modifying the ldscript to 

```
- RAM1 (xrw)                 : ORIGIN = 0x20000008, LENGTH = LD_MAX_DATA_SIZE - 8
+ RAM1 (xrw)                 : ORIGIN = 0x20000008, LENGTH = LD_MAX_DATA_SIZE - 4
```

No changes

federicaventriglia, 2025-02-12:
Yeah, apologies for the typo, looks like this 

```
/* Specify the memory areas */
MEMORY
{
FLASH (rx)                 : ORIGIN = 0x08000000 + LD_FLASH_OFFSET, LENGTH = LD_MAX_SIZE - LD_FLASH_OFFSET
RAM (xrw)                 : ORIGIN = 0x20000004, LENGTH = LD_MAX_DATA_SIZE - 4
RAM_SHARED (xrw)           : ORIGIN = 0x20030000, LENGTH = 10K
}
```

federicaventriglia, 2025-02-12:
Do you think the problem is in the clock configuration? Running the default one could also not work?

fpistm, 2025-02-12:
@federicaventriglia 
I've made the PR to support the generic.
It is untested as I do not have hardware with those mcu.

Some notes:
- Generic clock used only internal clock, so by default you can't use RF as it required an HSE.
- the value of FLASH and RAM have to be carefully reviewed as not all the FLASH nor RAM available (Flash/2 and RAM -64k)
- about the blink, the generic does not defined LED by default as it is a generic and so don't know which pin can be used. So defined like this:
```
#ifndef LED_BUILTIN
  #define LED_BUILTIN           PNUM_NOT_DEFINED
#endif
```
To redefine it to the proper value, create a [`build_opt.h`](https://github.com/stm32duino/Arduino_Core_STM32/wiki/Customize-build-options-using-build_opt.h) file with for example a LED connected to `PA5`:
`-DLED_BUILTIN=PA5`

federicaventriglia, 2025-02-12:
Thank you.

One question about the clock configuration, could you explain why there's this part?

```
 LL_PWR_SMPS_SetStartupCurrent(LL_PWR_SMPS_STARTUP_CURRENT_80MA);
 LL_PWR_SMPS_SetOutputVoltageLevel(LL_PWR_SMPS_OUTPUT_VOLTAGE_1V40);
 LL_PWR_SMPS_Enable();

  /* Select HSI as system clock source after Wake Up from Stop mode */
  LL_RCC_SetClkAfterWakeFromStop(LL_RCC_STOP_WAKEUPCLOCK_HSI);

```

my CubeMX generated file didn't have it and I wonder what it does. 

Also, the Flash and Ram sizes in the board_entry.txt file were wrong in the template; that could also be the issue.

fpistm, 2025-02-12:
> Thank you.

Welcome, let me know if the blink works.

> One question about the clock configuration, could you explain why there's this part?
>
> ```
>  LL_PWR_SMPS_SetStartupCurrent(LL_PWR_SMPS_STARTUP_CURRENT_80MA); [...]

IIRW, this is for the lowpower part and notmanaged by CubeMX.
You can find it as example in the project. I know it is not easy part.

> Also, the Flash and Ram sizes in the board_entry.txt file were wrong in the template; that could also be the issue.

No, it provides the full size but depending of the linker script, it have to be updated in the boards.txt when added.

fpistm, 2025-02-12:
OK. Let me know and I will merge the PR.

fpistm, 2025-02-13:
Yes it should be correct.

federicaventriglia, 2025-02-13:
When going from cubeIDE to arduinoIDE with the same MCU, same clock configuration I lose access to my board and have to perform a Under Reset chip erase, I wonder if it could be that the variant_generic files also need to be adjusted.

federicaventriglia, 2025-02-13:
If I disable USB CDC this doesn't happen, but I am able to create the Virtual Com Port in CubeIDE and use it.

fpistm, 2025-02-13:
Your clock config uses the HSE?
If yes what is the HSE value ?

federica-ven, 2025-02-13:
Yeah, that's the value I've set. This is the clock configuration I've validated with CubeIDE.

[image]

fpistm, 2025-02-13:
OK thanks.
Hard to help without the hardware nor debug session.
Could you share your cube project,please?
You can send it to my mail if you like.

federica-ven, 2025-02-13:
Would you be ok sharing your email?

fpistm, 2025-02-13:
frederic.pillon@st.com

fpistm, 2025-02-13:
Issue understood:
As the generic is used all pins from the `PinMap_USB` are initialized.
But PA13 is a jtag pin:
[image]
[image]

So while there is no reset pin PA13 is no more configured as SWDIO but as USB_NOE.

To fix this, as the array is a weak one, it can be redefined like this on the sketch to only init PA11 and PA12.
```
const PinMap PinMap_USB[] = {
  {PA_11, USB, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF10_USB)}, // USB_DM
  {PA_12, USB, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF10_USB)}, // USB_DP
    {NC,    NP,  0}
};

```

closed 2025-02-13 by fpistm

federica-ven, 2025-02-26:
Hi, still related to this Generic STM32WB board. 
I'm trying to enable the use of stm32duinoBLE and from what I understand I need to make some adjustments, like enabling the HSE clock. What I've done is configure BLE using CubeIDE and the STM32_WPAN library, following the official guide. I've also re-configured my clock and updated the firmware on the MCU using the STM32CubeProgrammer, using the BLE_HIC_layer file, as suggested from the guide (Firmware Update + Start Wireless Stack) 

I still seem to be unable to start the BLE service. When I try to run in the debugger the code seems to hand in the `reset()` part of of `BLE.begin()` at this line 

```
  _HCITransport->write(txBuffer, sizeof(pktHdr) + plen);
```

in `HCI.cpp` line 471. 

My clock config looks like this now (I've just placed in my main sketch for now, but it is being called (I've checked with breakpoints in the debugger)

```

extern "C" void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

  /** Macro to configure the PLL multiplication factor
  */
  __HAL_RCC_PLL_PLLM_CONFIG(RCC_PLLM_DIV8);

  /** Macro to configure the PLL clock source
  */
  __HAL_RCC_PLL_PLLSOURCE_CONFIG(RCC_PLLSOURCE_HSE);

  /** Configure LSE Drive Capability
  */
  HAL_PWR_EnableBkUpAccess();
  __HAL_RCC_LSEDRIVE_CONFIG(RCC_LSEDRIVE_MEDIUMHIGH);

  /** Configure the main internal regulator output voltage
  */
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);

  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI48|RCC_OSCILLATORTYPE_HSI
                              |RCC_OSCILLATORTYPE_LSI1|RCC_OSCILLATORTYPE_HSE
                              |RCC_OSCILLATORTYPE_LSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_ON;
  RCC_OscInitStruct.LSEState = RCC_LSE_ON;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSI48State = RCC_HSI48_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.LSIState = RCC_LSI_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_NONE;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }

  /** Configure the SYSCLKSource, HCLK, PCLK1 and PCLK2 clocks dividers
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK4|RCC_CLOCKTYPE_HCLK2
                              |RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_HSE;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.AHBCLK2Divider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.AHBCLK4Divider = RCC_SYSCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_1) != HAL_OK)
  {
    Error_Handler();
  }
}
```

Is there anything else that I need to account for in the variant configuration?

[wb55vg_ble_test.ino.zip](https://github.com/user-attachments/files/18982617/wb55vg_ble_test.ino.zip)

fpistm, 2025-02-26:
Hi @federica-ven 
this is not the right place for this as not related to the core nor the addition of generic. Discussion would be better and ease follow up.

I didn't see any hsem_lock, I'm not an expert but IIRW it is required.
ex:
```
  /* This prevents concurrent access to RCC registers by CPU2 (M0+) */
  hsem_lock(CFG_HW_RCC_SEMID, HSEM_LOCK_DEFAULT_RETRY);
...
  /* This prevents concurrent access to RCC registers by CPU2 (M0+) */
  hsem_lock(CFG_HW_RCC_SEMID, HSEM_LOCK_DEFAULT_RETRY);

...
 hsem_unlock(CFG_HW_RCC_SEMID);
```

Other point, is the RNG as stated in other WB clock config:
`  /* RNG needs to be configured like in M0 core, i.e. with HSI48 */
`

Could you try to mimic those one.

Some other point:
* Is the your HSE 32MHz? else it need to be redefined.
`#define HSE_VALUE    (32000000UL) /*!< Value of the External oscillator in Hz */`
* Did you flash the stm32wb5x_BLE_HCILayer_fw.bin to 0x080E0000?

fpistm, 2025-02-26:
> Can you elaborate a little on this? I'm not sure I understand what I need to change, if anything?
> 

As said try to mimic the clock config, this is part of the config:
https://github.com/stm32duino/Arduino_Core_STM32/blob/7095e16d33d34d08a01f640e26361099eb08ebb8/variants/STM32WBxx/WB55R(C-E-G)V/variant_P_NUCLEO_WB55RG.cpp#L147-L158

fpistm, 2025-02-26:
Seems logical, as when the cpu2 disable the RNG, it disables the HSI48 but if the clock ressource is lock thanks the hardware semaphore it prevents to disable the HSI48 also used by USB as input clock.
So your issue is resolved?

PR #2655 (2025-02-13): variants(wb): add generic WB55V(C-E-G) and WB55VYY
Fixes #2653.
commit 345138a9 (2025-02-13): variants(wb): add generic WB55V(C-E-G) and WB55VYY

Fixes #2653.