stm32duino/Arduino_Core_STM32 #2663: [New Variant] NUCLEO-L412RB-P

lylecheatham, 2025-03-01:
**Board description**
 - Name: NUCLEO-L412RB-P
 - Link: https://www.st.com/en/evaluation-tools/nucleo-l412rb-p.html

Hi there,

I'm willing to submit a PR for this but I'm a bit confused as to how! It seems that the guide in the wiki is for a "generic" board. Is a nucleo a generic board or how would I go about this? I'm switching a project I have over to this microcontroller and it's all Arduino code that I'd rather not port 😁 

I found this previous Issue and it made it seem like version 2.0 would have it but it does not seem to be the case https://github.com/stm32duino/Arduino_Core_STM32/issues/1350

If you just point me in the right direction then I'll get going!

fpistm, 2025-03-03:
Hi @lylecheatham 
Right, Wiki describes only the generic.
And no the 2.x does not provide this support, it was only mentioned in the issue that 2.0.0 have to be released before try to make a PR.
You can mimic any PR to see how  dedicated board is added.
Like this one:
https://github.com/stm32duino/Arduino_Core_STM32/pull/2634/commits/296fa63e29930afc4733bc7d28722f1056af795f

fpistm, 2025-03-04:
Added to  #722.

closed 2025-03-04 by fpistm