stm32duino/Arduino_Core_STM32 #2666: USB Client fails on power up on STM32C07

gus-ghielec, 2025-03-03:
Using Necleo64 with STM3232C07 support, I have this code.

```
void setup() {
  SerialUSB.begin();
}

void loop() {
  SerialUSB.println("Hello!");
  delay(500);
}
```

It works fine when loading from Arduino over DFU. However, the moment the system is reset USB will not work anymore. I suspecting an issue with USB clock initialization. USB works after DFU because USB clocks are already initialized but from a could boot USB Arduino does not work. Windows will say "unrecognized device" as teh deice is not responding to windows properly.

fpistm, 2025-03-03:
Hi @gus-ghielec 
add:
`  while(!SerialUSB);`

closed 2025-03-03 by fpistm

gus-ghielec, 2025-03-03:
Same behavior. This code works fine when loaded over DFU but then USB unrecognized after reset.

```
void setup() {
  SerialUSB.begin();
  while(!SerialUSB);
}
void loop() {

}
```

fpistm, 2025-03-03:
It works fine on my side using Nucleo G071.

fpistm, 2025-03-03:
Just remembered that C0 USB support is not in the 2.9.0 release. It has been merged in the main after the release.
https://github.com/stm32duino/Arduino_Core_STM32/pull/2584
You have to wait the 2.10.0 or use the main.

gus-ghielec, 2025-03-03:
Oh ok great, thanks. This is blocking issues so any estimate when can we continue the work? Where do I look to see if the release is out yet?

fpistm, 2025-03-03:
in the coming days/weeks. Arduino IDE will warn you that an update is available for the core.