stm32duino/Arduino_Core_STM32 #2694: [New Variant] Discovery kit STM32L562E_DK

haydenth, 2025-03-19:
- Name: STM32L562E_DK
 - Link to all relevant information:
   * [Documentations](https://www.st.com/en/evaluation-tools/stm32l562e-dk.html#documentation)
   * [Where this hardware can be purchased](https://www.st.com/en/evaluation-tools/stm32l562e-dk.html)

Onboard MCU: `STM32L562QEI6Q`

I'm happy to try to add this variant myself, but I'm not sure where to put it
```
~/.arduino15/packages/STMicroelectronics/hardware/stm32/2.8.1/variants/STM32L5xx $ ll
total 68
drwxrwxr-x 17 tom tom 4096 Mar 19 09:00  ./
drwxrwxr-x 22 tom tom 4096 Mar 19 08:59  ../
drwxrwxr-x  2 tom tom 4096 Nov 27 14:22 'L552C(C-E)(T-U)_L562CE(T-U)'/
drwxrwxr-x  2 tom tom 4096 Nov 27 14:22 'L552CE(T-U)xP_L562CE(T-U)xP'/
drwxrwxr-x  2 tom tom 4096 Nov 27 14:22  L552MEYxP_L562MEYxP/
drwxrwxr-x  2 tom tom 4096 Nov 27 14:22  L552MEYxQ_L562MEYxQ/
drwxrwxr-x  2 tom tom 4096 Nov 27 14:22 'L552Q(C-E)IxQ_L562QEIxQ'/
drwxrwxr-x  2 tom tom 4096 Nov 27 14:22  L552QEI_L562QEI/
drwxrwxr-x  2 tom tom 4096 Nov 27 14:22  L552QEIxP_L562QEIxP/
drwxrwxr-x  2 tom tom 4096 Nov 27 14:22 'L552R(C-E)T_L562RET'/
drwxrwxr-x  2 tom tom 4096 Nov 27 14:22  L552RETxP_L562RETxP/
drwxrwxr-x  2 tom tom 4096 Nov 27 14:22  L552RETxQ_L562RETxQ/
drwxrwxr-x  2 tom tom 4096 Nov 27 14:22 'L552V(C-E)TxQ_L562VETxQ'/
drwxrwxr-x  2 tom tom 4096 Nov 27 14:22  L552VET_L562VET/
drwxrwxr-x  2 tom tom 4096 Jan  9 10:23 'L552Z(C-E)TxQ_L562ZETxQ'/
drwxrwxr-x  2 tom tom 4096 Nov 27 14:22  L552ZET_L562ZET/
```

I am guessing this is the right location: `L552Q(C-E)IxQ_L562QEIxQ` - but just looking for advice on this. Also, since it's a DK is there a recent DK that anyone has added I can use as a template? Thanks

haydenth, 2025-03-19:
Got it. Thank you. Most of it is straightforward but I'm having trouble getting the `SystemClock_Config(void)` function right, I think. Is there guidance somewhere on how to define this?

fpistm, 2025-03-19:
> Got it. Thank you. Most of it is straightforward but I'm having trouble getting the `SystemClock_Config(void)` function right, I think. Is there guidance somewhere on how to define this?

For the generic or for the DK?
What is your problem "to get it right"?

haydenth, 2025-03-19:
Just the DK, but I think I answered my own question already - sorry!

PR #2696 (2025-03-28): [STM32L5xx] Add support for STM32L562E-DK board
Adds support for the STM32L562E-DK Discovery Kit

- Included new board configuration in README.md
- Updated boards.txt with STM32L562E-DK details
- Added variant files for STM32L562E-DK
- ldscript.ld for the STM32L562E
- Modified generic_clock.c for compatibility with STM32L562E-DK

Resolves #2694
closed 2025-03-28 by fpistm