stm32duino/Arduino_Core_STM32 #2736: [New Variant] WeAct STM32H503 Core Board

Avamander, 2025-05-24:
**Board description (please complete the following information):**
 - Name: WeAct STM32H503 Core Board
 - Link to all relevant information: https://github.com/WeActStudio/WeActStudio.STM32H503CoreBoard/tree/master
 - Schematic: https://github.com/WeActStudio/WeActStudio.STM32H503CoreBoard/blob/master/Hardware/WeAct-STM32H503Cx_CoreBoard_V10%20SchDoc.pdf

**Note that user can add a STM32 based board following this [wiki](https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-(board))**

Any contribution is welcome, so do not hesitate to submit a PR.

All new variant request will be moved to [New Variant] request list #722 and the related issue will be closed.

Avamander, 2025-05-24:
For some reason I also can't get USB CDC serial working with the following configuration generic STM32H503CBTx, CDC (Generic Serial), U(S)ART (Generic Serial). The device fails to respond with a descriptor and the result is "This device cannot start. (Code 10)" under Windows 11. At the same timeDFU works and enumerates properly on the same board.

fpistm, 2025-05-24:
You should check the clock configuration to ensure USB clock is 48 MHz.

Avamander, 2025-05-24:
Reading this, it should be: https://github.com/stm32duino/Arduino_Core_STM32/blob/main/variants/STM32H5xx/H503CB(T-U)/generic_clock.c

It also doesn't seem like USB communication is failing completely, if I fiddle with the clocks then it fails sooner and the error is also about failing to enumerate (not just "This device cannot start.").

fpistm, 2025-05-26:
USB is know to work on Nucleo H503RB with generic or dedicated target. I do not have this hardware so can't help on this.

Moved to [New Variant] request list #722.

closed 2025-05-26 by fpistm