stm32duino/Arduino_Core_STM32 #2743: GPIO 15 aka PB6 does not work as an output for STM32U385

alexceltare2, 2025-05-29:
**Describe the bug**
When uploading the Blink code either via Arduino IDE or Cube, D15/PB6 does not light up when I attach a LED to it. Not sure if it's by design or there is a hardware bug.

**To Reproduce**
Upload the Blink program and change LED_BUILTIN to 15

Steps to reproduce the behavior:
1. Open the Blink program.
2. Change LED_BUILTIN to 15 or PB6

**Expected behavior**
LED should light every second.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Arduino IDE version: 2.3.5
 - STM32 core version: 2.10.1
 - Tools menu settings if not the default: 3.3V, Mass storage
 - Upload method: USB embedded ST-Link

**Board (please complete the following information):**
 - Name: Nucleo-U385RG-Q
 - Hardware Revision: N/A
 - Extra hardware used if any: N/A

fpistm, 2025-05-29:
Gpio15 is not PB6. It is GPIO pin 15 from one GPIO port. 
PB6 is GPIO PIN 6 from GPIO PORT B.

alexceltare2, 2025-05-29:
Nucleo-U385RG-Q

closed 2025-05-30 by fpistm

fpistm, 2025-05-30:
It works. What is your code?

alexceltare2, 2025-05-30:
Nevermind, it works now that I've switched to 3.3V from 1.8V. Thanks for the help.