stm32duino/Arduino_Core_STM32 #2759: Missing entry for STM32L052K8Ux in boards.txt

jakezimmer, 2025-06-29:
**Describe the bug**
The boards.txt entry for the L052K8Ux is missing but the L052K8Tx exists as a menu option. Is there a reason this wasn't included or should I add it in and submit a pull request? There looks to be a [variant folder](https://github.com/stm32duino/Arduino_Core_STM32/tree/main/variants/STM32L0xx/L052K(6-8)U_L062K8U) for this board but no entry. I'm designing a board using this chip and want to make sure there isn't a reason it cannot be used with the Arduino Core before moving forwards.

**To Reproduce**

Steps to reproduce the behavior:
1. Choose the "Generic STM32L0 series" for board.
2. Look at board part numbers.

**Expected behavior**
There should be an option for this part number.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Arduino IDE version: 2.3.4
 - STM32 core version: 2.10.1

**Board (please complete the following information):**
 - Name: Generic STM32L0 series

fpistm, 2025-06-29:
Hi  @jakezimmer 
There is no issue. This mcu has not been added yet. 
You cam add it: https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-%28board%29

fpistm, 2025-07-01:
Added to #722