stm32duino/Arduino_Core_STM32 #2766: analogRead returns 0 if serial is disabled on variant F030F4 HSE 8MHz

r-pv, 2025-07-08:
Hi all.

Using the stock variant for the F030F4 HSE 8MHz, when using analogRead always returns 0 if the usart is disabled, if the option "enabled (generic serial)" is selected the analogread returns the expected values, even if serial is not used.

Tested this behavior on pins PA0, PA1 and PA2.

I'm using an hal_conf.extra.h with disabled stuff:

```
//#define HAL_ADC_MODULE_DISABLED
//#define HAL_I2C_MODULE_DISABLED
#define HAL_RTC_MODULE_DISABLED
#define HAL_SPI_MODULE_DISABLED
//#define HAL_TIM_MODULE_DISABLED
#define HAL_DAC_MODULE_DISABLED
#define HAL_EXTI_MODULE_DISABLED
#define HAL_ETH_MODULE_DISABLED
#define HAL_SD_MODULE_DISABLED
#define HAL_QSPI_MODULE_DISABLED
```

I'm using Arduino IDE 2.3.5 and the stcore 2.10.1

r-pv, 2025-07-08:
Hi @fpistm !

Tried your code (w/ slight modification for the led pin):

also with the PA0 connected to VCC/3.3V

```
const byte led=PA4;

void setup() {
	pinMode(led, OUTPUT);
}

void loop() {
	int sensorValue = analogRead(A0);
	if (sensorValue != 0) {
		for (uint8_t idx = 0; idx < 5; idx++) {
			digitalWrite(led, HIGH);
			delay(100);
			digitalWrite(led, LOW);
			delay(100);
		}
	}
	delay(1000);
}
```

tried normal, no debug-env, with serial disabled I got no led activity.
with serial enabled got led activity.

with debug-env and serial disabled, got led activity even when the debug-env didn't run as expected:

[image]

what mcu are you using, the led blinks if you aren't using the debug-env?

thanks!.

fpistm, 2025-07-08:
I see no related stuff in the errata sheet. you best bet is to debug. You need a stlink and enable the "Optimize for debugging" in the sketch menu.