stm32duino/Arduino_Core_STM32 #2770: Errors on compilation of "dac_write_value"

beaka, 2025-07-11:
&#x26A0; When reporting any issue, please try to provide all relevant information to help on its resolution.

**Describe the bug**

`/home/bruce/.arduino15/packages/STMicroelectronics/hardware/stm32/2.10.1/cores/arduino/wiring_analog.c: In function 'analogWrite':
/home/bruce/.arduino15/packages/STMicroelectronics/hardware/stm32/2.10.1/cores/arduino/wiring_analog.c:196:7: error: implicit declaration of function 'dac_write_value'; did you mean 'adc_read_value'? [-Wimplicit-function-declaration]
  196 |       dac_write_value(p, ulValue, do_init);`

**To Reproduce**
compile some code.

Steps to reproduce the behavior:
Menu: Sketch, Verify/Compile.

**Expected behavior**
No library errors to show up.

**Desktop (please complete the following information):**
 - OS: Fedora 42
 - Arduino IDE version: 2.3.6
 - STM32 core version: 2.10.1
 - Tools menu settings if not the default: [e.g. Newlib Standard, No Serial]
 - Upload method: DFU

**Board (please complete the following information):**
 Have not got far enough to purchase one.

fpistm, 2025-07-11:
Hi @beaka .
Could you be more precise.
Which target ? Option? Sketch?
Because there is no issue to build any target else I guess I would have several issue....

beaka, 2025-07-11:
The target was Generic STM32H7/Gerneric H747AGIx

Default options for selecting the board. Just want to see if any code will compile, will upload whenever I can get any code to compile.
So far have not been able to get any code downloaded other than some platform examples to compile. 

I am trying to see if there is any better way for ADC/DAC use than analog read. Do not want to use CubeIDE. I did use libmaple years ago to sample at a specific sample rate and output to PWM but that will not work in V2 Arduino IDE.
I have tried F4xx and G474 HAL code downloaded from various places and none compiles either.

[main.zip](https://github.com/user-attachments/files/21184722/main.zip)

fpistm, 2025-07-11:
This is known to compile.
Maybe you get an weird install. Could you try to uninstall and install it again. My guess is the installation was corrupted.