stm32duino/Arduino_Core_STM32 #2783: Hardware Serial Halfduplex does not recieve anymore after sending - reintilaization with Serial.begin(xx) required

helicocrasher, 2025-07-31:
**Describe the bug**
For a slave device on a polled half duplex serial bus, I wrote some code. 
Here the operation principle: 
1.  The bus master device sends start character plus a  slave address. 
2.  The addressed slave device sends a 8 byte data packet onto the bus and waits for being polled the next time. 

The code of my slave device correctly answered to the first polling sequence by sending a perfect 8 byte packet.
Then waited for more received data, but Serial2.available() never turned true again, despite lots of traffic on the bus.
Workarounds: 
a) Not using half duplex and externally tying RX and TX pins together worked. 
b) Using half duplex Serial2.setHalfDuplex() and and restarting the USART with Serial2.begin(57600) after every transmission of an 8 byte data packet also made the code work perfectly.

Question is this a bug or is this like half duplex is supposed to work? I guess bug.

**To Reproduce**
Code - happy to share on request

Steps to reproduce the behavior:
1. Plug
2. Press USER_BTN
3. See error

**Expected behavior**
No serial restart required after every packet sent also in half duplex mode

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Windows 11 Pro Build 26100
 - VS Code 1.102.3 - PLatformIO core 6.1.18 home 3.4.4
 - STM32 core version: 2.11 (directly from github)
 - Tools menu settings if not the default: NA
 - Upload method: SWD ST-Link V2

**Board (please complete the following information):**
 - Name: WeAct STM32G031F8P6  / genericSTM32G031F8
 - Hardware Revision: V1.0
 - Extra hardware used if any: FrSky Archer Plus GR6 as bus master, SMart Port Sniffer on a 2nd STMG031 board

**Additional context**
On request

helicocrasher, 2025-07-31:
Hi @fpistm 

Thank you for the fast reply!
No I did not use the enableHalfDuplexRx(). 
I looked at the linked code. Does this mean that sending data is automatically disabling the reception of data when in half duplex mode? And hence requires an explicit enable command? Some kind makes sense and I completely missed that part.
It would explain the observed behavior.

I will modify my code and replace the Serial2.begin() in the loop code with Serial2.enableHalfDuplexRx() and let’s see whether it works - I expect it will.

Will report the results when tested.
:-)

fpistm, 2025-08-01:
Fine. Thanks for the feedback.

closed 2025-08-01 by fpistm