stm32duino/Arduino_Core_STM32 #2794: readAnalog() on Nucleo F746ZG problem

xRadx, 2025-08-19:
Hi,
as in the subject
it seems that while using the above board - analogRead(PA3) returns value of approx 3k all the time, regardless of the voltage supplied to A0 pin.

But the same code works correctly on Nucleo F767ZI board.

fpistm, 2025-08-19:
Hi @xRadx 
I've tested and it works as expected.
Using `analogRead(A0);` or `analogRead(PA3);`

[image]

fpistm, 2025-08-19:
Are you sure you select the right Board part number in the menu?

xRadx, 2025-08-19:
Yes, I am sure

fpistm, 2025-08-19:
Could you tell me what is the board version? You should have a sticker with mbxxxxxx