stm32duino/Arduino_Core_STM32 #2814: Generic F411CEUx digitalPinToPinName() bug

dfleck, 2025-09-06:
It seems `digitalPinToPinName()` is slightly broken on Generic F411CEUx. Version 2.11.0 on Arduino 2.3.5 on Mac.

Also on BlackPill FC411CE

`Serial.println(PB_13_ALT1);` returns 285
`Serial.println(digitalPinToPinName(PB_13_ALT1));` returns 286

Discovered the problem trying to do:
`SPIClass  SPI_4(PA1, PA11, PB_13_ALT1);`

fpistm, 2025-09-06:
No issue here. You just not use correct syntax.
Thus is not

closed 2025-09-06 by fpistm

dfleck, 2025-09-06:
So the problem is not in `digitalPinToPinName()` but must be somewhere in the pin definitions. Investigating further:

In variants/STM32F4xx/F411C(C-E)(U-Y)/PinNamesVar.h 
`PB_13_ALT1 = PB_13 | ALT1,` so PB_13_ALT1 is a valid syntax

`Serial.println(PB_13_ALT1);` prints 285
`Serial.println(PB13_ALT1);` prints 284
`Serial.println(PB_13 | ALT1);` prints 285
`Serial.println(PB_13);` I think should be 28 but it prints 29

Could it be off-by-one because there's no `PB_11` in the `const PinName digitalPin[] = {` table in `variant_generic.cpp`?

fpistm, 2025-09-08:
> PB_13_ALT1 = PB_13 | ALT1, so PB_13_ALT1 is a valid syntax
Yes it is a valid syntax but you mixed type between pin number (PYn) and PinName (PY_n), they are not the same.
All Arduino API uses pin number PYn is an alias to ease use of STM32.
In your case with generic F411CEUx:
`PB13` is a pin number with value 28. It is used as index to find the PinName `PB_13` in the digitalPin array.
So `PB_13` is equal to ` PB_13 = (PortB << 4) + 0x0D` with `PortB = 0x01`  then `PB_13 = 0x1D = 29`
`ALT1 = 0x100`. 
So no issue here you only mix type.
`digitalWrite(PB13)` or `digitalWrite(28)` or `digitalWrite(D28)`is correct but `digitalWrite(PB_13)` is not correct as you will write in fact  to pin number 29 `digitalWrite(29)` which is `PB14`.