stm32duino/Arduino_Core_STM32 #2816: EEPROM.h Fails on STM32G474RBT6 due to Incorrect HAL_FLASHEx_Erase Parameters

ShaggyDog18, 2025-09-10:
**Describe the bug**
The standard EEPROM.h library fails to perform EEPROM emulation on the STM32G474RBT6 microcontroller. The behavior has two distinct symptoms:
- On a fully erased chip: The very first EEPROM.put() or EEPROM.write() call only succeeds in writing the first 4 bytes of data. All subsequent bytes in the data structure remain 0xFF.
- On subsequent resets: All EEPROM.put() calls fail silently. The data in flash is never updated, making it impossible to change a value after the initial partial write.

The root cause appears to be that the EEPROM.h library is passing incorrect parameters to the low-level HAL_FLASHEx_Erase() function for this specific MCU variant (when configured in single-bank mode). A direct call to the HAL function with the correct parameters works perfectly, proving the hardware and HAL drivers are functional.

**To Reproduce**
Complete source code which can be used to reproduce the issue:

This standard "boot counter" sketch demonstrates the failure. It uses the EEPROM.h library as intended.
`#include <Arduino.h>
#include <EEPROM.h>

// A simple struct to test with
struct MySettings {
  uint32_t bootCount;
  uint32_t someOtherValue;
};

void setup() {
  Serial.begin(115200);
  while (!Serial);
  delay(2000);

  Serial.println("\n--- EEPROM.h Library Test on STM32G474RBT6 ---");

  MySettings settings;
  
  // Read the struct from emulated EEPROM
  EEPROM.get(0, settings);

  Serial.print("Data read from flash: bootCount = ");
  
  // Check if the value is 0xFFFFFFFF (erased state)
  if (settings.bootCount == 0xFFFFFFFF) {
    Serial.println("[First Run / Erased]");
    Serial.println("Initializing new settings...");
    settings.bootCount = 1; // First boot
  } else {
    Serial.println(settings.bootCount);
    Serial.println("Settings loaded. Incrementing for next boot...");
    settings.bootCount++; // Increment the boot count
  }

  // Attempt to write the updated struct back. This is the operation that fails.
  EEPROM.put(0, settings);

  Serial.println("Write operation called.");
  Serial.print("New value for bootCount should be: ");
  Serial.println(settings.bootCount);
  
  Serial.println("---------------------------------------------");
  Serial.println("Press RESET to test the next cycle.");
}

void loop() {
  // Do nothing.
}`

Steps to reproduce the behavior:

- Use an STM32G474RBT6 board (custom or otherwise). Ensure it is configured for Single-Bank Mode via the DBANK option byte in STM32CubeProgrammer.
- Upload the sketch above.
- Open the Serial Monitor at 115200 baud.
- First Run: Observe the output. It will correctly identify the first run and report the new boot count as 1.
- Press the RESET button on the board.
- Second Run (See error): Observe the output again. Instead of reading 1 and incrementing to 2, it will read the old value from the first run (1) or a garbage value, because the EEPROM.put() call failed to erase the flash page. The boot counter will never increment past the first write.

**Expected behavior**
[... 128 lines cut ...]
  if (settings.bootCount == 0xFFFFFFFF) {
    Serial.println("[First Run]");
    settings.bootCount = 1;
    settings.calibrationValue = 3.14f;
    strcpy(settings.deviceName, "MyG4Device");
  } else {
    Serial.println(settings.bootCount);
    settings.bootCount++;
  }

  Serial.print("New bootCount is: ");
  Serial.println(settings.bootCount);
  Serial.println("Writing back to flash...");

  EEPROM_write(&settings); // Write using our custom function

  Serial.println("Write complete. Please RESET to test again.");
}

void loop() {
  delay(5000);
}``

This strongly suggests the EEPROM.h library is constructing the FLASH_EraseInitTypeDef struct with incorrect parameters (most likely the .Banks field) when targeting the STM32G474RBT6.
**Note: troubleshooted with help of Gemini 2.5 Pro**

ShaggyDog18, 2025-10-01:
yes, will test shortly

ShaggyDog18, 2025-10-01:
These settings do work! Thank you!
So, the configuration is for a single bank, Bank#1:

`-DFLASH_BANK_NUMBER=1 
-DFLASH_END=0x0801FFFF 
-DEEPROM_FLASH_PAGE_NUMBER=31`

fpistm, 2025-10-02:
Fine, thanks for the feedback.
So mainly configuration issue. In 99 % of the case EEPROM config is correct. In specific case, it requires to have custom user configuration when the default FLASh config is changed.

closed 2025-10-02 by fpistm